cleanflight/blackbox-log-viewer #4: In Firefox, Playback stops when trying to jump to a certain spot in the graph

Arakon, 2015-04-19:
When attempting to click the graph below the main window in Firefox (Windows 7, 37.0.1), most of the time, playback stops as if it was reset to the beginning and paused, the red bar disappears, and playback can not be resumed. Seemingly at random, it does work to skip to the spot in the graph, but only rarely and not the same spot twice. Also, the index timer in the bottom right displays NaN:NaN:NaN when this happens.
[image]

None of this occurs with Chrome, using the exact same log file and exact same log viewer.

thenickdude, 2015-04-22:
Thanks for the report, this should be fixed now. Please try the latest version and let me know if it still doesn't work!

closed 2015-04-22 by thenickdude

Arakon, 2015-04-22:
Indeed, appears to be working just fine now.

commit 60523372 (2015-04-22): Fix clicking on seekbar on Firefox failed

Closes #4