cleanflight/blackbox-log-viewer #7: Add ability to render and export video

Mauronic, 2015-04-23:
Title says it all. Would be a killer feature!

joshuabardwell, 2015-06-20:
Seconded. Currently, I use a screencast app to record playback, but this isn't ideal for any number of reasons.

IIRC, @thenickdude previously said that the main issue is ability to render video from Java.

thenickdude, 2015-06-21:
Yeah JavaScript doesn't really have a video encoder available. The closest thing I've found is the [Whammy WebM encoder](https://github.com/antimatter15/whammy), which is a hack that only works on Chrome. This would need significant reworking in order to make it suitable for blackbox video rendering.

fedorcomander, 2015-06-21:
Can we use host application via messaging? Yes, its in chrome only.

fedorcomander, 2015-06-21:
Another option would be generate ffmpeg script for the video file, log index, dimensions, log start...

joshuabardwell, 2015-06-21:
The idea I had was that possibly the features from the viewer could be cross-ported to the renderer, then the viewer could be set up to export a config file that could be read by the renderer, which would then generate the PNG overlays from the data. That way you could use the awesome UI of the viewer to set up your traces instead of cumbersome command line options.

It would be great if the viewer and renderer could be coded so as to easily share feature sets, but given that the renderer was not developed in JS and the viewer is, I suspect this would be challenging.

fedorcomander, 2015-06-21:
While ago i made script to render from command line, if you interested.

See video description https://www.youtube.com/watch?v=bBzVdYO5u04

May be you can use it for a while / change it a bit for your needs.

joshuabardwell, 2015-06-21:
@fedorcomander this is fine, but AFAIK, BB render doesn't support the customizable graphs that BB viewer does.

closed 2016-09-19 by thenickdude