cleanflight/blackbox-log-viewer #14: Ability to zoom in the timeline

borisbstyle, 2015-06-10:
It would be nice to be able to change reporing time interval to see more clearly differences between loop to loop and smaller ripples or even to zoom out to see the bigger picture.

ledvinap, 2015-06-10:
Using blackbox_decode and then processing data in some other program will be easier if you are interested in detailed analysis ...

borisbstyle, 2015-06-10:
Yes I know, but having this ability in log viewer would be nice.

fedorcomander, 2015-06-10:
Not sure its what requested, but might be handy:

[image]

fedorcomander, 2015-06-10:
And zoom in.

[image]

borisbstyle, 2015-06-10:
@fedorcomander 
Yes that's exactly what I meant!

hydra, 2015-06-10:
Oops wrong button

thenickdude, 2015-06-11:
Merged.

closed 2015-06-11 by thenickdude