cleanflight/blackbox-log-viewer #16: Allow user to select graph line thickness

hydra, 2015-06-10:
With the laptop in the field on a sunny day i find it very hard to see the lines.  Thicker lines would help.

thenickdude, 2015-06-11:
Until a configuration dialog for this is added, I've added a PLOT_LINE_WIDTH constant so you can tune it manually in the source:

https://github.com/cleanflight/blackbox-log-viewer/blob/master/js/grapher.js#L19

timdonovanuk, 2016-03-20:
Changing this constant doesn't seem to do anything for me. Have tried values 0.5, 0.9, 5, 50, 100.

timdonovanuk, 2016-03-20:
It seems whatever value you put in the constant its immediately overwritten on line 665 with plotLineWidth:             Math.max(1.25, canvas.height / 400)

If you want to make the line thicker edit line 444,  canvasContext.lineWidth = <a value>

McGiverGim, 2017-07-17:
This was done long time ago. I close the issue.

closed 2017-07-17 by McGiverGim