cleanflight/blackbox-log-viewer #20: Add ability to customize graph color

joshuabardwell, 2015-06-20:
I prefer my graph colors to be consistent, so that I can learn to intuitively read the graphs. For example, I always set my graphs up with P, then I, then D so that the colors are consistent. If I have P and D turned on but I turned off, I set the traces up as 1:P, 2:D, 3:D, so that the 3rd trace overlays the 2nd trace and maintains the same color as if the 2nd, I trace, was on. This is kind of cumbersome. Ideally, I'd like to be able to assign a color to each of the traces manually and have that setting remembered. For example, in a master configuration, assign PID_P[pitch] = cyan across all graphs; PID_I[pitch]=magenta; and so forth.

thenickdude, 2015-06-22:
I'm leaving this open as an enhancement because user-configurable colours would be neat, but I'm adding bug #22 because I think that the current automatic colour assignment is poor enough to be considered a bug. It could be improved markedly even before user customisable colours are added.

McGiverGim, 2017-07-17:
@joshuabardwell @thenickdude This issue is too old. I think that this is already in the current version. Is that true? Can this issue be closed?

closed 2023-04-19 by hydra