cleanflight/blackbox-log-viewer #25: Error opening log: No logs in this file could be parsed successfully

canphaz, 2015-06-30:
Hi, on both blackbox_decode and the blackbox-log-viewer, i am getting the following message when i open one of the .TXT files from my Blackbox SD card:

Error opening log: No logs in this file could be parsed successfully

I can't seem to get the tool to accept any of the files, yet they are a few hundred kilobytes big.
Thanks.

[image]
[image]
[image]

thenickdude, 2015-06-30:
That log file is actually MSP data, did you connect the logger to the wrong serial port? Post your ports configuration.

canphaz, 2015-06-30:
Ahh that was it, I only had the checkbox "MSP" checked for UART1, after checking "Blackbox" checkbox, it now logs properly :) thanks a lot, sorry for the noob question, didn't realize i needed to do that as well.

closed 2015-06-30 by canphaz

thenickdude, 2015-07-01:
No problem!