cleanflight/blackbox-log-viewer #26: Can't play log but log is loaded

deeph46, 2015-07-21:
Hi,

I want to look my log in viewer but I can't play it in my Chrome. Sometimes shows WebGL error. I can load the log into viewer but the viewer shows me blank, the graph don't exist.
I'm using Windows 8.1 32bit, Ati Radeon Xpress 1100.

I Googled for WebGL error and give me result that is from VGA driver. I have updated it but still no luck. Any how to fix this?

Thanks

thenickdude, 2015-07-22:
When you use the Cleanflight Configurator, does the 3D model of the craft on the main page show as a simple box (rather than a quadcopter model with propellers and everything)? If so, your graphics card or graphics card drivers don't support the WebGL that the log viewer currently uses for its 3D craft model. In this case I can fix the log viewer to use a 2D craft model instead.

deeph46, 2015-07-22:
In the Cleanflight Configurator the 3D model show as a quadcopter model. Then if using 2D craft model, does it work like 3D craft model (like the motors movement)?.

thenickdude, 2015-07-22:
Interesting, in that case your graphics card does support WebGL. Can you email me your log file? n.sherlock@gmail.com

deeph46, 2015-07-23:
Sent.

deeph46, 2015-07-23:
Sorry, just noticed that in the Cleanflight Configurator the 3D model is in box mode.

thenickdude, 2015-08-17:
Ah okay, in that case your computer doesn't support the 3D model. I've now added a fallback mode to the log viewer which will draw the craft in 2D on your computer instead, please update to the latest version:

https://chrome.google.com/webstore/detail/cleanflight-blackbox-expl/cahpidddaimdojnddnahjpnefajpheep

closed 2015-08-17 by thenickdude