cleanflight/blackbox-log-viewer #31: unable to open logfile in viewer

lichtl, 2015-08-28:
I currently struggle to open a blackbox logfile within the viewer. I am able to extract it using the tools but there seems to be no interaction with the viewer once I try to open it. Please see below the decoding (it might be possible that entry 2 is causing this) and the attached logfile

blackbox_decode LOG00134.TXT
Decoding log 'LOG00134.TXT' to 'LOG00134.01.csv'...

Log 1 of 4, start 00:10.546, end 01:25.804, duration 01:15.257

Statistics
Looptime           1882 avg          458.0 std dev (24.3%)
I frames    2312   38.1 bytes avg    88195 bytes total
P frames   34570   24.5 bytes avg   846409 bytes total
E frames       1   12.0 bytes avg       12 bytes total
S frames      46    4.0 bytes avg      184 bytes total
Frames     36882   25.3 bytes avg   934604 bytes total
Data rate  490Hz  12467 bytes/s     124700 baud

17 frames failed to decode, rendering 115 loop iterations unreadable. 38360 iterations are missing in total (38367ms, 50.98%)

Decoding log 'LOG00134.TXT' to 'LOG00134.02.csv'...

Log 2 of 4Statistics
E frames       1   12.0 bytes avg       12 bytes total
Frames        0
Data rate: Unknown, no timing information available.

1 iterations are missing in total (0ms, 100.00%)

Decoding log 'LOG00134.TXT' to 'LOG00134.03.csv'...

Log 3 of 4, start 02:25.534, end 02:29.398, duration 00:03.864

Statistics
Looptime            999 avg           23.2 std dev (2.3%)
I frames     121   33.7 bytes avg     4074 bytes total
P frames    1812   21.8 bytes avg    39527 bytes total
E frames       2    9.0 bytes avg       18 bytes total
S frames       1    4.0 bytes avg        4 bytes total
Frames      1933   22.6 bytes avg    43601 bytes total
Data rate  500Hz  11579 bytes/s     115800 baud

1932 loop iterations weren't logged because of your blackbox_rate settings (1931ms, 49.99%)

Decoding log 'LOG00134.TXT' to 'LOG00134.04.csv'...

Log 4 of 4, start 03:16.295, end 03:27.295, duration 00:11.000

Statistics
Looptime            999 avg           23.5 std dev (2.4%)
I frames     338   34.4 bytes avg    11642 bytes total
P frames    5043   22.6 bytes avg   113946 bytes total
E frames       2    9.0 bytes avg       18 bytes total
S frames       5    4.0 bytes avg       20 bytes total
Frames      5381   23.3 bytes avg   125588 bytes total
Data rate  489Hz  11548 bytes/s     115500 baud

3 frames failed to decode, rendering 14 loop iterations unreadable. 120 iterations are missing in total (119ms, 1.09%)
5500 loop iterations weren't logged because of your blackbox_rate settings (5499ms, 50.00%)

[image]

thenickdude, 2015-08-29:
This behaviour will be improved in the next release of the log reader with b4d7fb682507e9d585f9c6ddd02d8303b928f18e, which allows your 3rd and 4th flights to be viewed. The issue is triggered by heavily truncated headers in your log which is rendering the log unreadable. I hope to improve this header logging issue with https://github.com/cleanflight/cleanflight/issues/1270

thenickdude, 2015-08-29:
Logging reliability in this situation (fast looptimes) should now be improved with the next release of Cleanflight since it will include this fix:

https://github.com/cleanflight/cleanflight/commit/2b356a47dccf814690f393f27b6fadb5f719519e

Another thing you can try is ensure that you format your microsd card using the full erase option of this SD card formatting tool:

https://www.sdcard.org/downloads/formatter_4/

That allows some writes to the card to complete faster, which increases logging reliability, particularly if your card has been filled with data some time in the past.

lichtl, 2015-08-29:
Thanks. I will also give the full format SD card a try.

closed 2015-08-29 by lichtl