cleanflight/blackbox-log-viewer #46: log truncated error

giova-fr, 2016-06-27:
Hi !

I have no issue to read my log files with blackbox_render tool.
But when i try to load them into  blackbox explorer then it says : "Error : Log is truncated, contains no data"

I record with those settings : 
115200,26,3,0,1,1,0
baud,escape,esc#,mode,verb,echo,ignoreRX

Blackbox is triggered from a switch.

And I import my log files directly from the SD card (i cannot plug openlog TC to my FC to import from cleanflight).

May you please tell me what is going wrong?
i've attached a log as example.
[LOG00016.zip](https://github.com/cleanflight/blackbox-log-viewer/files/334894/LOG00016.zip)

samfok, 2017-06-30:
Which firmware version are you using?
I was using Naze firmware v2.0.3 and ran into the same issue where blackbox_decode and blackbox_render process log files just fine but the blackbox_log_viewer reported "Error : Log is truncated, contains no data". However, [beta-flight blackbox explorer](https://chrome.google.com/webstore/detail/betaflight-blackbox-explo/canpiagfkeefejklcojkhojdijglnghc?hl=en) was able to render the data. Using pre v2 Naze firmware (v1.14.2) to record blackbox data, the cleanflight blackbox explorer was able to render blackbox data.

I figure there must have been a switch in blackbox data format between firmware versions?

McGiverGim, 2017-07-01:
Can you try manually installing the master version of cleanflight blackbox explorer? I think this issue must be solved.

samfok, 2017-07-01:
Indeed there is no issue when running the master version of the cleanflight blackbox explorer. I had originally installed the blackbox explorer via the chrome web store, which is going to be retired anyways since chrome is moving away from apps. To reduce the number of users who run into this issue, the installation instructions could be updated to drop the chrome web store method.

closed 2023-04-19 by hydra