cleanflight/blackbox-log-viewer #47: Blackbox explorer freeze at the calculating screen

sfersystem, 2016-08-17:
Hi, 

I'm using this tool and works very well but when i try to render a logfile without video, i have the screen that say calculating and it freeze at this point. I can move the windows, the app is working but the render won't start.

I'm under windows 10, the app is installed by chrome.
thank you.

thenickdude, 2016-08-17:
Can you please email me the log file that you're trying to render?
N.sherlock@gmail.com

thenickdude, 2016-08-17:
Actually, never mind, I can reproduce that with my own logs at the moment,
I'll look into it, thanks!

thenickdude, 2016-08-17:
This has now been fixed in version 0.1.6.2. Chrome should update it automatically over the next day, but you can update manually by going to Chrome Menu -> More Tools -> Extensions, ticking the "developer mode" tickbox, then clicking the "update extensions now" button that appears. You should see the version number next to "CleanFlight - Blackbox explorer" update to 0.1.6.2 after a minute or so.

closed 2016-08-17 by thenickdude