cleanflight/blackbox-log-viewer #49: Exporting video always Calculating

apeelme, 2016-09-18:
I tried on Debian Jessie Mate, Ubuntu Mate 16.04 (now fresh install) & Chrome 53.0.2785.116 (64-bit) at all times and nothing happens, always Calculating... the file size, and no rendering.

Here is the DeveloperTools console output:
http://imgur.com/a/eypT0

thenickdude, 2016-09-18:
This is a bug in Betaflight's branch of the log viewer which has already
been fixed in Cleanflight's version (the one in this repository). Please
use their bug tracker to report it to them.

Cheers,
Nick

apeelme, 2016-09-18:
I installed from the Chrome WebStore (CF & BF BlackboxExplorer) and both have this problem. Here's the CF console: http://imgur.com/a/sT6lb

apeelme, 2016-09-18:
Download this repo...load index.html, open log file, export video...same error. :(

thenickdude, 2016-09-19:
Ah okay this is apparently a WebP encoding quirk specific to Chrome 53, because it doesn't seem to happen any longer on Chrome 54. It should now be worked around in the newest version, 0.1.6.4. You can update to this version manually by ticking the "developer mode" tickbox on Chrome's extensions settings page, then clicking the "Update extensions now" button.

closed 2016-09-19 by thenickdude