cleanflight/blackbox-log-viewer #66: View MAVLink Log Files

semi10, 2017-08-23:
Hello! 
I want to add the ability to use blackbox-log-viewer with MAVLink .bin (log files)
Currently i think about some kind of log file converter from MAVLink .bin files to blackbox .txt
Is there some relevant projects/papers/blog post that i should read?
I found good [description ](https://github.com/cleanflight/cleanflight/blob/master/docs/development/Blackbox%20Internals.md) of Blackbox protocol. Should I learn it or it will be enough to use included libraries? 
Can you suggest better approach? Any help will be welcomed! Thx in advance!

McGiverGim, 2017-08-23:
After the description doc that you already have found, the best way to know how it works is to take a BB file and debug it in the log viewer. You will see how the fields are readed. 
Opening a BB file with some editor and take a look to the contents will help too.
I don't know anything about the MAVLink log files, so I don't know if it has an equivalent data to the BB file. If it has, then your converter maybe can work ;)

semi10, 2017-08-23:
>  the best way to know how it works is to take a BB file and debug it in the log viewer
I'll take a look. Thank you.

closed 2017-08-25 by mikeller