cleanflight/blackbox-log-viewer #74: Chrome Extension Version and Releases are all Messed-up

spatzengr, 2017-12-03:
See https://youtu.be/dnqv-sP-AtE.  What is the scoop?  Why is the latest version of Chrome (2.5.13) have less feature, more missed header file reads, and the scaling of RC command screwed up and 2.5.11, which you can download here, seem better and RC command scaling is right?

McGiverGim, 2017-12-04:
The actual version was started again from the betaflight one. After that, missing changes were added. Maybe some of them were lost.
If you know the commit/change of the missing feature I can try to add it again.

spatzengr, 2017-12-06:
The code in the Master here is right (https://github.com/cleanflight/blackbox-log-viewer   --- version 2.5.11).  How does the Chrome App get updated?  The Chrome app. that is currently version 2.5.13 the scaling is off and it seems like an old version of Blackbox.  How do we update the Chrome app. to be the version of BB that is download under the codes tab of this GitHub project (cleanflight/blackbox-log-viewer).

mikeller, 2017-12-06:
@spatzengr: @hydra does the releases to the Chrome web store.

spatzengr, 2017-12-06:
Does Boris do for Betaflight?  They both need a good updating from the latest repositories.  CF is way old, and BF says it is version 2.5.13, but is something else.

closed 2023-04-19 by hydra