cleanflight/blackbox-log-viewer #87: Standalone app in Firefox does not work

sanchosk, 2019-08-16:
Firefox seems not to be able to run the blackbox-log-viewer as standalone app.
Steps to reproduce:
1. Install firefox quantum 68.0.1 (64-bit) on Ubuntu
2. Download full zip of blackbox-log-viewer from github and unzip it to a folder
3. Run firefox and open the index.html page
4. See debug console:
`20:06:57.789 require does not exist, maybe inside chrome index.js:84:17
20:06:57.788 ReferenceError: chrome is not defined main.js:1892:9
20:06:57.792 ReferenceError: chrome is not defined release_checker.js:16:5
20:06:57.809 ReferenceError: chrome is not definedlocalization.js:9:9
`
Try to open any BBL file, nothing will happen.

McGiverGim, 2019-08-16:
You don't need to install it in this strange way, you have a .deb package in the releases page.

sanchosk, 2019-08-17:
Thanks for the help! This worked, although it did not want to open 11.8MB log, just the smaller ones (4MB was OK).

closed 2019-08-17 by sanchosk