cleanflight/blackbox-log-viewer #88: Separate the log parsing code out into a reusable library?

qdrk, 2021-12-27:
Hi there, a long term follower of cleanflgiht blackbox log analysis here. I'm wondering if folks are fine to sparate the parsing code in flightlog_parser.js out into a reusable library that can be used by third-party apps? The reason for the ask is, with the current logging logic in the firmware, especially with all the field encodings and frame format, it's really hard to code something from ground-up (nor does it make much economic sense) just to work with the logging code that was written over 5 years ago. Hopefully with a separate library, it'll make things easier and encourage a richer ecosystem around PID tuning and issue diagnosis, as other developers can focus on high-level data representations instead of worrying about all the nitty-gritty encoding/frame format issues.

fedorcomander, 2021-12-27:
there is pretty good c bb decoder available

---
Cheers,
Alex

fedorcomander, 2021-12-27:
But yeah, js lib would be nice to have (read/write). I literally needed one yesterday! Just bb-decoded it to csv and did the rest :)